ranenbg/Arduino-FFB-wheel #87: About ffb texture

laurolopesdantas, 2025-02-13:
Hi Milos, me again bothering you, I did some tests with the ffb and the only thing I could find that I dislike is the texture of the scrub effect while braking, feels a little bit odd, not mute, but different of anything else I tried before, is like i can't feeling that signature bouncing texture on the ffb, is that something that I could tweat with the sliders of configuration, or mess up with the code frequencies to tweak that ?

ranenbg, 2025-02-13:
You can adjust PWM settings, to give you a little bit more details, because this is what defines the final FFB resolution, regardless of what the game is sending.

I made safe defaults in firmware phase correct, 16kHz, pwm+-, which gives +-500 steps of FFB resolution in both directions. You can use fast top, 8kHz this wilo give you +-2000 steps of FFB which should be enough.

Make sure you don't use more than 100% in general gain and constant force sliders, as this lowers the FFB resolution as well. Use FFB monitor to fine tune level of FFB strenght in game such that you barely see red (avoid clipping).

Ultimately, FFB quality depends on a game, car, setup, track..

laurolopesdantas, 2025-02-13:
Doesn't change, but as you said I notice that is game related, in AC the scrub ffb is pretty much the same I'm use to, but in AMS 2 has this odd and different feeling, I suspect that is more advanced, in others games the scrub feels like a kurb, but in ams2 is much more subtle, like a sandpaper, I think I wold have to go in ther other way and tune down de details on ffb to get the same result of AC, again thanks.

ranenbg, 2025-02-13:
Note that you have to restart arduino in order to apply any changes in pwm settings.

Each sim has its own "flavour" of FFB, that we get so much used to after playing for a long time, in a way that everything else just feels wierd. But the truth is that none of sims have "real" FFB, because in a real car most of the car feel comes from the seat and g-forces, the forces in the wheel alone are not telling much. Game devs have to incorporate all of that into wheel forces in some way. Each one has their own idea how that should be done and that is why each sim has its own charm.

laurolopesdantas, 2025-02-13:
yeap, you're right. Now i'm experiencing suddenly lost of ffb mid turn, seems the spring just gave up and return a few seconds latter, this is a known issue ?

ranenbg, 2025-02-14:
No, I never noticed loss of ffb just all of a sudden. In which game is this?

laurolopesdantas, 2025-02-14:
In AMS2, using a custom ffb profile

ranenbg, 2025-02-14:
Ok, I have that game, will check.

In the meantime, could you please describe how can I reproduce ffb loss behaviour?

laurolopesdantas, 2025-02-14:
I just select custom ffb(I will send you the file) gain 50, fx 50 everything else at 0, then load interlagos on f3 301 version 5 laps and that's it, after a few laps the loss of ffb starts to happen mid turn, then return after a few seconds or turns, as I said is like the spring just gave up suddenly

[ffb_custom_settings.txt](https://github.com/user-attachments/files/18800628/ffb_custom_settings.txt)

[image]

laurolopesdantas, 2025-02-18:
Hi again, I noticed the same problem in AC, so is not game related, then I tried everything, change USB, change the config, use a note shared gnd to the BTS and nothing, even suspecting of overheating in the motot or BTS, so watching the ffb monitor on AC I noticed that was clipping and staying in the red zone for too long and Tried to reduce de gain in game, that seems to help, do not know yet if solve for complete the problem, but definitely help, there is a way to debug the program to know whats happening when the problem show up ?

ranenbg, 2025-02-18:
I have the ffb debugging feature in the firmware. It is possible to see exactly which ffb effects the game is sending and with which parameters.

I was suspecting a problem with periodic effects which have timing, but you said it's also in AC. Those games only send constant force ffb effect and eventualy damper effect. Both of these are time independent, which leads me to beleive there is a hardware issue on your side. If ffb monitor still shows signals when you observe ffb cut, that means it's not firmware related.

Bts has internal over temperature protection, where it shuts down. It could be this.

laurolopesdantas, 2025-02-18:
Yeah, In think that could be that, the ffb monitor keeps running normal and sometimes in the red when the problem happens, also the BTS heatsinc was very hot, I'm gonna try some cooling solutions, thanks

ranenbg, 2025-02-18:
As long as the heatsinc is hot that is good, because you can always move heat away from it by directed airflow from a fan for example.

laurolopesdantas, 2025-02-18:
Today is a very hot day here in Brazil and placing a fan righ on the BTS seems to be solved or mitigated the problem