ranenbg/Arduino-FFB-wheel #88: Hall effect sensors for pedals

MrCjxer, 2025-02-17:
Hi bro

Can i use this hall effect sensors for pedals?🤔

[image]
[image]
[image]
[image]

ranenbg, 2025-02-17:
Hi bro. I've looked ar the datasheet for 3144, it seems that is only has a switching operation, meaning only 2 states. Unless you want to use your pedals like buttons, this will not work. LM394 is just an opamp which sets the high level of signal to 5V I guess. In other state when there is no mag field, I suppose output is 0V.

MrCjxer, 2025-02-18:
Hi
Yes i did tested it and it's work just like a button
But what model do you recommend to me ?
I bought 3 of these but i should throw it to trash 😭

ranenbg, 2025-02-18:
Sorry bro, S49E is analog.

MrCjxer, 2025-03-18:
Thank you bro

closed 2025-03-18 by MrCjxer