ranenbg/Arduino-FFB-wheel #89: Arduino not recognized

namcktv, 2025-02-20:
Windows has stopped this device because it has reported problems. (Code 43)

A request for the USB device descriptor failed.

The arduino back to normal when i upload another sketch not your project, i think its a problem with the hex file

namcktv, 2025-02-20:
Please compile brWheel_my.ino.leonardo_v241chfx.hex and brWheel_my.ino.leonardo_v240zfx.hex and brWheel_my.ino.leonardo_v240fx.hex

ranenbg, 2025-02-20:
Tnx man, I'll try to do it as soon as I can.

ranenbg, 2025-02-21:
Hi, I have just uploaded the existing HEX files in my GitHub repository you mentioned to my Leonardo board and this is what I found:
[1] leonardo_v240zfx.hex - reported and works normally
[2] leonardo_v240fx.hex - reported and works normally
[3] leonardo_v241chfx.hex - reported and works normally

I don't keep the source code for v23X firmware so I can't recompile its HEX anymore (you should use v24X), but anyway, I gave them the same checking and I found that:
[1] leonardo_v230zfx.hex - reported and works normally
[2] leonardo_v230fx.hex - reported and works normally
[3] leonardo_v231chfx.hex - reported and works normally

So everything seems fine from my end. You should download the HEX from my repository, don't get it from the release. I've re-compiled some HEX after the release as there were some configuration mistakes. I'm preparing a new release with some updates and that is why I didn't re-release v24X.

namcktv, 2025-02-22:
My Arduino Leonardo may be faulty, I tried things related to i2c and serial port all stuck not booting and Windows doesn't recognize arduino, thanks for the help. By the way I may do a project for STM32F4XX based on your project, can you support me?