ranenbg/Arduino-FFB-wheel #91: How to use the hardware wheel re-centering feature?

shubham0x13, 2025-03-15:
I am using an AS5600 sensor and need guidance on how to use the hardware wheel re-centering feature. How does it work, and what steps should I follow?

ranenbg, 2025-03-15:
Hi, unfortunately, in the current version of firmware it's not possible to use an i2C bus device like AS5600 and centering button at the same time. They both require hardware interrupt capability of pin D2. For i2C bus this is the data line or SDA and centering button is also setup on this pin.

Tnx very much for asking about this, as it didn't occur to me before. I'm at the moment working on a new firmware version where I have made quite some changes. One of those is a possibility to completely disable usage of an optical encoder - its two hardware interrupt capable pins TX and RX become unused. I'll have a look if there is enough memory left to remap the center button pin to one of those two if optical encoder is not used and if as5600 is used. I think it will be doable.

shubham0x13, 2025-03-15:
Thanks for the clarification! I appreciate the effort you're putting into the new firmware version with added flexibility. Looking forward to the update, great work!

ranenbg, 2025-03-15:
You're welcome dude, tnx for this great suggestion. After thinking about how to implement it, I'm very optimistic, since it's only a matter of remaping center button pin, by using compiler predefinitions. No actual changes in the code are neccessary.

shubham0x13, 2025-03-15:
Maybe a dumb question, but I need to connect a switch/button to a pin and design my DIY wheel in such a way that it gets triggered when the wheel is in the centered position.

ranenbg, 2025-03-15:
No, it's much simpler than that.

You will connect a button or a switch to a center button pin in a way that once you press it it connect the center button pin to the ground, that's all.

The firmware is using interrupt and a pullup. I made it in such a way that you firat need to press the button, then once you release it the centering function is executed.

shubham0x13, 2025-03-15:
Ohh, I see! I initially thought it would automatically calibrate the wheel's center position at boot or on a button press, similar to an endstop in 3D printers. But now I get it pressing the button simply brings the wheel back to the center position I calibrated earlier.

ranenbg, 2025-03-15:
It's even simpler than that. There is nothing to calibrate, I simply initialize as5600 position at 0deg on arduino powerup. So you have two options:

[1] align wheel to center, plug arduino to usb port
[2] If it happens that your wheel was angled before arduino startup you will move it back to center by hand and then press center button

shubham0x13, 2025-03-15:
Got it! Thanks for the explanation and help.

MrCjxer, 2025-03-18:
> Tnx very much for asking about this, as it didn't occur to me before. I'm at the moment working on a new firmware version where I have made quite some changes. One of those is a possibility to [...]

Hey buddy. What?! Did I hear that right?! You want to completely disable optical encoder support?! 😲
Or do you want to add the ability to disable the optical encoder in the firmware? 😃

MrCjxer, 2025-03-20:
> Nothing to worry about. It's going to be just a firmware option, where if you comment out the usage of the quad encoder in Config.h simply the functions for the encoder will not be used and some program memory will become free.
> 
> update. I have already implemented this feature in new firmware, the re-center button is remapped to the TX pin, when no optical encoder is used and when the magnetic encoder is used along side center button firmware option. I still have plenty of things to finish up and test before I can release v250, but it's coming soon.

OH , thank you bro . poof .

And when will this new firmware be released?  
If the exact date is not specified, give me an estimate because I want to plan accordingly, my friend. 😊🔥

ranenbg, 2025-03-20:
I plan to release it as soon as it's ready.

I have finished coding firmware and GUI, updated documents, manuals, pinouts, and wiring images. I still have to compile HEX files for both Leonardo and proMicro. There are new firmware options, so now I have many more combinations for HEX, I'm still deciding which ones to compile. After that, I have to upload everything to Git Hub and update the readme's. I don't want to give any rough estimates, but most of the hard work I have already done, what's left is the most boring stuff.

MrCjxer, 2025-03-20:
Hi

Oh yeah, thanks bro! 😎😍
I wish you success, both in work and in life. 😊🙌🏻🥂

ranenbg, 2025-03-20:
I've just uploaded a new firmware release. The original question is implemented, so I will close this issue.

closed 2025-03-20 by ranenbg