ranenbg/Arduino-FFB-wheel #92: Custom Firmware Modifications & Feature Requests 🚀

MrCjxer, 2025-03-20:
# 📌 Custom Firmware Modifications & Feature Requests 🚀
# 📌 Custom Firmware for Racing Setup 🚀  

## Project Overview  
Hello, my friend! 😊  

I've been trying to modify a firmware for my setup, but due to my **limited programming knowledge**, I haven't been able to implement the changes I need, even after months of effort. I would really appreciate your help!  

---  

## 🔧 Hardware Configuration  

### 🎛️ Analog Inputs (A)  
| **Pin** | **Function** |  
|---------|------------|  
| **A0**  | Throttle   |  
| **A1**  | Brake     |  
| **A2**  | Clutch    |  
| **A4**  | Button (or disabled due to a faulty board) |  
| **A5**  | Empty     |  

### 🎛️ Digital Inputs/Outputs (D)  
| **Pin** | **Function** |  
|---------|------------|  
| **D0**  | Encoder   |  
| **D1**  | Encoder   |  
| **D2**  | Re-center wheel |  
| **D3**  | Empty     |  
| **D4**  | Empty     |  
| **D5**  | Empty     |  
| **D6**  | NANO 16 Button |  
| **D7**  | NANO 16 Button |  
| **D8**  | NANO 16 Button |  
| **D9**  | Motor Driver |  
| **D10** | Motor Driver |  
| **D11** | Motor Driver |  
| **D12** | Empty     |  
| **D13** | LED / Empty |  

---  

## 🎮 Requested Modifications  

| **Pin** | **New Function (Convert to Button)** |  
|---------|--------------------------------|  
| **A4**  | Button (or disabled) |  
| **A5**  | Button |  
| **D3**  | Button |  
| **D4**  | Button |  
| **D5**  | Button |  
| **D12** | Button |  
| **D13** | Button (if LED is not used) |  

## ⚙️ Additional Features Required  

| **Option** | **Function** |  
|-----------|------------------------------------------|  
| `"a"`     | Enable pedal axis auto-calibration (otherwise, manual calibration) |  
| `"c"`     | Hardware wheel re-center support |  
| `"h"`     | Hat Switch (D-pad) support |  

---  

## 📂 Preferred Output  
✅ **Open-source code** (preferred)  
✅ Otherwise, a **HEX file** would work  

💰 I understand that this requires time and effort. If necessary, I am **willing to compensate** for your hard work with a payment or donation. I'd rather support **you** than go to someone else.  

**I truly appreciate your help!** Looking forward to your response. 😊🙏

ranenbg, 2025-03-20:
You may check out HEX v250achexn, it should have everything you mentioned.

MrCjxer, 2025-03-22:
> You may check out HEX v250achexn, it should have everything you mentioned. For more details refer to ver_info_leonardo_micro.txt

Hi
Thank you for your response but bro for example on " E " option " clutch " is unavailable ...

Please can you do this for me ?!
I'm looking for someone do it for me and i want pay to him
I can donate it to you or i pay to you
Thank you ❤️

ranenbg, 2025-03-22:
I've already invested a lot of my time on v250 release. I need some motivation..

MrCjxer, 2025-03-22:
Yeah i know 🥲
It's hard to do it just yourself
But I'll wait a bit so you can rest and regain your motivation.

I'm from Iran, and the financial situation is very tough.
I can send you half of my total balance, which is about $5.
Is that okay?
My whole balance is 9$ :)

ranenbg, 2025-03-22:
No need to send anything, keep your balance where it is. It's weekend, I'm resting and spending time with my family.

MrCjxer, 2025-03-22:
❤️

MrCjxer, 2025-03-22:
> No need to send anything, keep your balance where it is. It's weekend, I'm resting and spending time with my family.

Oh sorry bro 🙌🏻
I wish you and your family best 💗

ranenbg, 2025-03-24:
I've fixed the ffb issue. You may try HEX v250achxn.

MrCjxer, 2025-03-27:
Hi
Yes i tested new firmware and FFB works fine 👌🏻
But i have found another problem with FFB which i will open another issues or here 

**"About the buttons - are you saying it's absolutely impossible to have 22 buttons?** 🤔  
When using a button box (Arduino Nano), can I really not use the main board's pins for buttons? ( A5, D3, D4, D5, D12, D13 are empty ) 

I need:
- 8 for gears
- 12 on the steering wheel
- 2 shifters behind the wheel"

ranenbg, 2025-03-27:
Hi looking forward to your new issue about FFB.

Well, you are right, 22 buttons is possible, I never said it's not. I just didn't structure my code that way, so it's not a small task to get it to work and I was avoiding it so far. You are the first one who needs nano button box + buttons on leonardo at the same time. I'll add it to my to do list and see what can be done.

MrCjxer, 2025-03-27:
1 - Regarding the FFB issue
In Assetto Corsa, when I'm going in reverse and want to turn quickly, if I rapidly turn the wheel left or right, the wheel itself spins 3 or 4 times and exceeds the 900-degree rotation I set for it.

2 - And yes, I know XY shifter can be used, but my shifter is designed for micro switches. Changing it would take a lot of time and cost, otherwise it would be a great solution .

-

> I just didn't structure my code that way, so it's not a small task to get it to work and I was avoiding it so far. You are the first one who needs nano button box + buttons on leonardo at the same time .

3 - So from what I understand, it's not something I or another programmer can add as a feature.
It's all on you, buddy. ♥

**4 - I think it would be good if you could make it so that any extra pin on the main board can be converted into a button. This way, I think fewer pins would be wasted, right?**

ranenbg, 2025-03-27:
1. The FFB issue you're talking about is not caused by my firmware, it's how the game handles forces. There is very little that can be done in firmware to prevent it. The only thing I could do is to set ffb to 0 wheel axis goes past the range, but I'm not liking this idea - once the wheel goes back inside range a max force will be slammed all of a sudden.
2. I understand, that's fine.
3. Anyone can do it if enough effort is invested.
4. Yes, that would be great so as milion of other things that I would like this firmware to have. The complexity is that I have already many configurations or firmware options that change the firmware in many ways. It has become so difficuilt to manage all possible combinations simply by using precompiler defines. I will see what I can do.

MrCjxer, 2025-03-27:
1 - So this is a game issue, but do commercial wheels like Logitech and Moza have the same problem?

2 - Yeah I know damn it, it's really problematic.
But you could make these features available only to those who compile the code themselves.
I mean in the **" config.h "** file where we add or remove // comments.
At first I thought it was difficult, but after doing what was suggested, I realized it's actually quite simple - just by commenting or uncommenting lines.
This was just a suggestion, but I hope you can do it

MrCjxer, 2025-03-27:
WOW ! Dude, you're a genius!

After all this time searching for a 9-pin cable and socket, or looking for a way to add 6 extra buttons with 16 buttons on leonardo, then boom, discovered this shift register trick!!
**" r - support for external shift register chips for 24 buttons (3x SN74ALS166 wired in series) "**

1- Does it have to be exactly this model of shift register?
Because I couldn't find this exact model to purchase from any store

2 - Can shift registers be used with the H (Hat Switch) option as well?

3 - Is it possible for you to create a visual guide (like you did for other components) for using:

3.1 - A single shift register

3.2 - Three shift registers

I believe this would solve my problem - and many other users who need more buttons - eliminating the need to convert those 6 pins or other pins into button inputs.

ranenbg, 2025-03-27:
Logitech gear driven wheels can't have such problem because they have a mechanical hard stop limit at end of axis range.

And for dd wheels like moza I don't know really. What keeps them inside of range is electrical endstop spring ffb effect. If you are stong enough amd if motor has low enough holding torque you can always push through it.

You wheel probably has a week motor and this electrical end stop is not so strong.

MrCjxer, 2025-03-28:
> You guessed it well, option "r" is for 3×8bit shift register chips that allow 24 buttons, but I was not able to get it to work yet. One has to use that SN74 chip, I got it from ebay. Anyway, there was some problem with arduino not creating stable enough clock, it was jumping all over the place and was not able to get the correct readings from shift register chips - no button readout at all. I simply gave up from it at that point as I had it working correctly when I used externaly generated clock from a pulse generator and used a scope to verify if chips were shifting bits, and they were. It was obvious, that these chips just require a stable clock to operate and arduino couldn't provide it with my implementation of "software" spi communication.
> .
> .
> .

1 - So you're saying I should wait until the 3 shift registers get fixed?  
Or is there a chance this might never work at all?  
I remember one user managed to make about 64 buttons with the same Arduino and shift registers.  
Why not ask them for help?

> .
> .
> .
> One has to use that SN74 chip, I got it from ebay.

**2 - Do you mean I can use any shift register that is SN74?  
Even SN74HC165N? Or only and only SN74ALS166?  
Because in Iran, we don't have "Ebay" to buy from 🥲**  

> .
> .
> .
> Your wheel probably has a week motor and this electrical end stop is not so strong. i wouldn't worry too much about it. My firmware keeps applying ffb even if you go past axis range. End stop force should move the wheel back inside the range.

3 - Yes, I know there's an end-stop force, but my exact issue is that at that specific moment I mentioned, it seems inactive and lets me push past it! And when I cross the end point, I have to manually rotate the wheel 2-3 full turns to get back into the force feedback range!  
Maybe it's a game bug—I need to test it again.  

4 - The motor I'm using is an RS 775 (brushless), and my wheel is wooden and super lightweight, so no power is wasted. In fact, when I tested the end-stop force feedback, it felt really powerful and satisfying—I loved it! 😅😍

MrCjxer, 2025-03-28:
Sorry, my mistake...
If you can , answer me
Thanks ❤️

ranenbg, 2025-04-02:
I've just published a new release to address this question. I think that we can now close this issue. If you have further questions make a new issue.

closed 2025-04-02 by ranenbg