ranenbg/Arduino-FFB-wheel #94: FFB Not Working In-Game After Updating to v250cdwhx

shubham0x13, 2025-03-22:
I was previously using v240wx, but after flashing v250cdwhx (along with the new GUI app), the FFB only works for idle spring force. However, in-game or when testing with WheelCheck, it doesn't work. Reverting to v240wx it works fine.

ranenbg, 2025-03-22:
Hi, tnx for report. which game did you use?

shubham0x13, 2025-03-22:
Assetto Corsa, F1 2020. In Assetto Corsa, when the car is idle, the wheel feels stiff, but when I accelerate, crash, or drive, it becomes loose.

ranenbg, 2025-03-22:
Assetto corsa uses damper ffb effect when car is stationary. You can try to lower damper gain slider in gui and see if that helps.

shubham0x13, 2025-03-22:
[image]
FFB only works for Centering Gain in the GUI app. When I open Wheel Check and select any effect, nothing happens.

ranenbg, 2025-03-22:
Ok, I'll have a look what is going on.

Would be good if you could check other HEX with "w" option. If they all show the same behaviour then just keep using 240wx untill I try to sort it out.

shubham0x13, 2025-03-22:
I tried v250acdwhx, but it's the same. Switching back to v240wx works.

ranenbg, 2025-03-22:
You can also try something with "b" and "w", that one has 2 ffb axis.

shubham0x13, 2025-03-22:
At first, I mistakenly tried using hex with "b" and thought that was why it wasn’t working. Later, I tried it without "b".

ranenbg, 2025-03-22:
That is extremely strange, I'll check it out.

MrCjxer, 2025-03-22:
Hi, yeah, I have this problem too.
I tested it in Assetto Corsa (it's the only game I have).

As long as I'm in the GUI and Windows, it works fine,
but as soon as I enter the game, it stops working.

The only thing that works is the last lock, which engages when the wheel reaches the end.

ranenbg, 2025-03-24:
I have found the issue and fixed it, it was something very minor. I've re-compiled all HEX and re-released v250 firmware. You may now check it out.

shubham0x13, 2025-03-24:
It's working now.

closed 2025-03-24 by shubham0x13