ranenbg/Arduino-FFB-wheel #95: Snowrunner

NinjaFrozr, 2025-04-01:
Can't get this to work in SnowRunner. There are in-game options for direct input mode and custom wheel preset but i still can't bind any axis, the game doesn't detect it.  Is there any way to make it work ?

ranenbg, 2025-04-03:
Hi, one thing that comes to my mind is that the game cannot detect the axis you're trying to bind because other axes are slightly moving due to ADC noise, and the game struggles with it. If that's the case, make sure to set axis calibration correctly such that absolutely no movement is seen if you're not touching anything. In some games, for the axis to be correctly detected, you must move it almost through its entire range, like more than 70%.

If this doesn't help, there is nothing else that can be done from my side regarding firmware.

shubham0x13, 2025-04-03:
Try https://www.youtube.com/watch?v=vEcKw1PVPpQ

shubham0x13, 2025-04-03:
> Is there any way to make it work ?

Try https://www.youtube.com/watch?v=vEcKw1PVPpQ