ranenbg/Arduino-FFB-wheel #96: Question about shift registers and Hall effect sensors

MrCjxer, 2025-04-13:
**Hi there,**

I'm glad the issue with the shift registers is resolved and we can now move forward 😊

I have a few questions:

### 1. Shift Register Compatibility  
Do we need to use the exact models you mentioned — **SN74ALS166N** or **SN74HC165N**?  
Unfortunately, these specific parts are not available in my country (Iran), but I have access to similar 8-bit shift registers, such as:  
- 74LS166AN  
- 74F166N  
- SN74166 (8-bit)  
- 74LS166  
- ...or similar variants.

👉 **Would any of these be acceptable substitutes?**  
Also, I’d really appreciate it if you could provide some guidance on how to choose the right shift register for this project.

### 2. Hall Effect Sensor Alternatives  
Can I use a model like **SS49E** as a Hall effect sensor alternative?

### 3. Recommended Hall Effect Sensors  
If possible, could you please list a few Hall effect sensor models that would be suitable for this project (for pedal use)?

Thanks so much for your time and support! 🙏

ranenbg, 2025-04-13:
Hi, you're asking me too much. You can find answers to all of those questions through some googling.

Any shift register with similair features as the two I mentioned will do, just look at the data sheet.

Any hall sensor with analog voltage output will do fine, like ss49e, those are the most standardly used. Just take care about correct magnet placement - two mag poles have to rotate around the sensor in order to have linear mapping between angle and voltage.

MrCjxer, 2025-04-13:
Hi. Yes i know bro 🥲
And I'm sorry for that ...

Believe me, I've been researching during these past two weeks since you uploaded the new firmware, but since my field isn't electronics, the more I research, the more confused I get.

Thanks for your guidance.

---

I managed to find and order the SN74HC165N version.
Could you please add a wiring diagram for this type of shift register on the main page as well? (Whenever you have time.)
Because, as always, it's confusing 🥲

---

And also, do Hat Switchs exist in this type of button box?
If they do, which pins are they connected to?
The first 4 pins of the first shift register maybe ?

---
I'm soooo sorry again bro

I’ll try to make this my last question, and after this, I’ll vanish into oblivion, never to return 🥲💔

ranenbg, 2025-04-13:
The information in the wiring diagram I already provided is enough to connect any shift register. Important part is that output of 1st one goes into arduino, and output of any shift register behind goes into the serial input of the one ahead - they are stacked in series.

Yes, hat switch will use first 4 bits, so that is pins A, B, C , D from 1st shift reguster chip.

MrCjxer, 2025-04-13:
Ok
Thank you so much ❤️🙏🏻
I'm sorry to waste your time :)

---
But I have a new idea.
It's about head tracking! ( Tracking the head in games :) )
I've done some research on it and I have some fairly good information.
If you're interested, let me know where you'd like me to explain it

Here, or by email, new issues or even if you're not interested in knowing at all.

Thank you again
And bye 👋🏻

ranenbg, 2025-04-14:
No thank you, I've played with camera head tracking some 20 years ago.

MrCjxer, 2025-04-14:
No, it's not camera Head tracking.
Bmi160 and MPU-6050 sensors and may other sensors.

I just wanted to share it with you in case you’d like to use it in this project or maybe a new one.
Thanks a lot!
Take care!

closed 2025-04-14 by MrCjxer