ranenbg/Arduino-FFB-wheel #97: can this use printer encoder

Gandoun, 2025-04-21:
can this use printer encoder

ranenbg, 2025-04-21:
Hi, short answer - yes.

Long answer - the firmware requires 2 channel quadrature encoder signals. Any device that can provide such signals will work, the only requirement is 5V logic level. Open collector encoders will work also, but you have to put a pullup resistor.

Gandoun, 2025-05-11:
[image]

when i press ok the application closes

shubham0x13, 2025-05-11:
Go to the `data` folder within the app directory, create a file named `COM_cfg.txt`, write your Arduino Leonardo COM port in it (e.g., `COM6`), save the file, and start the app.