ranenbg/Arduino-FFB-wheel #100: Shifter calibration

kick-a-poo, 2025-06-07:
Hello, thanks for this awesome program I have been working on it for a good 2 years, currently having an issue thaat might seem dumb, for some reason I cant calibrate the shift 

[image]

much help would be appreciated

ranenbg, 2025-06-07:
Hi, yes, there was a misshap from my side in the latest release. Just use the HEX with xy shifter option from the github repository, there I have fixed this and re-uploaded, as I didn't think it was worth it to make a new release just for 2-3 new HEX files.

ranenbg, 2025-06-07:
That's great. Would be very useful to make an "issue", and explain your wiring with dual bts, motor and gear/pulley setup, ratios.. Just plsease name it non-an-issue or something not to be missleading.

perides, 2025-09-18:
[image]

Can someone help me? I’m using the latest version of the firmware and also the wheel control, but I’m not able to calibrate the shifter.

I’ve tried everything:

- Using the latest precompiled versions of both
- Building and flashing from the source code

But in both cases I couldn’t get the calibration to work.

Thanks in advance.

ranenbg, 2025-09-19:
Great. I'm closing this thread as issues with shifter calibration are fixed.