ranenbg/Arduino-FFB-wheel #101: 250~~~~ issues I've been facing

Notriatv, 2025-06-09:
Hi Rane, greetings!

I’ve been testing the new version for a while, and here’s what I found a bit problematic:

The FFB feels weaker than in previous versions...
I tried a few variations of your new firmware version (v250xxxx), and I noticed that the FFB seems weaker compared to the older versions, even using the same FFB settings in the software. Sometimes, the force feedback doesn’t work at all in tools like rfrWheel Test or WheelCheck with this firmware.

Also, my analog H-shifter isn’t working… I mean, the analog movement is recognized by the software, but the buttons don’t turn green when they reach the activation zone.

Those are the little issues I’ve found so far.

Another point: version 241chfx, which I currently use (and thank you for that!), is still the best for my personal setup. It offers all the basic analog axes for a regular car, 16 buttons via the Nano, a working H-shifter, and strong, consistent FFB. I missed those features in the new versions.

The closest I found was v250hfxnl, but I’m not really interested in a load cell brake at the moment. I also missed the centering button.

So, since the FFB feels off and the H-shifter isn’t working properly, I’ve gone back to the 24xx firmware.

One more tip, in case you weren’t aware: the MT6701 magnetic encoder already works as a native replacement for the optical encoder — no special code needed. I’ve tested it, and it works great at 4096 CPR.

If you plan to keep developing this project and release future versions, please consider making something closer to the 241chfx, if possible. 🙂

Anyway, thanks so much in advance for all the hard work you’ve already put into this project!

ranenbg, 2025-06-16:
Hi, you are welcome, I'll try to keep this project alive and going as long as I can.

About weak FFB, I don't know why you feel that, I did not change the strength of FFB in new firmware. I did change the default PWM settings from 16kHz, 500 steps, phase correct to 8kHz 2000 steps and fast pwm. If you are using min torque slider above zero, there may be some differences so make sure that PWM settings are the same as for the ones you used in earlier firmware.

I'll see what I can do with compiling equivalent HEX for 241chfx. In the new firmware, this should be something like 250chnfx and should be doable concerning memory constraints.

Notriatv, 2025-06-27:
Hi Rane, I’m back. Thanks for making this firmware available — it fixed the H-shifter, but the FFB still feels weird... kind of stuck, “rubbery,” or sluggish.

And before you make any recommendations, I should say that I already tested with another Leonardo/BTS7960/motor, double-checked all the wiring or methods, and tried different settings in the GUI — but it still feels very different compared to version 240 or below
It’s like it's skipping steps or something — hard to explain.
Anyway, version 240 is already excellent for me.
The only change I’d like to know if it’s possible: maybe using the SN74 instead of the Nano to get more buttons.
Other than that, I honestly don’t see how this version could be improved.Not that I’m trying to limit your creativity, but considering the limitations of the Arduino Leonardo, this is already amazing!

Regarding the MT6701, I was having issues with it losing center — something that was solved by using shielded cables and a 1nF ceramic capacitor between VCC and GND

ranenbg, 2025-06-27:
The thing is that I can't go back to firmware 240, as this has been updated to v250, I can only add features to that one. I have already made SN74 for 24 buttons feature, it's only a matter of uncommenting it in the config.h. Let me know if you need me to compile you a new v250 HEX with your previous options, including the shift register chips. It can be from 1 to 3 chips (for 8 to 24 buttons).

Notriatv, 2025-07-03:
https://youtu.be/oW1KTgZWjRk?si=q0ugjlhHk6mlLqwk 
I tried making a video in English — hope it's understandable haha

Notriatv, 2025-07-03:
> The thing is that I can't go back to firmware 240, as this has been updated to v250, I can only add features to that one. I have already made SN74 for 24 buttons feature, it's only a matter of [...]

If it were possible to test an unofficial version of 240 with SN74 support, that would be great!
Right now, version 250 is just not usable for me due to these issues...
But no pressure — only if it’s not too much trouble for you. :)

Also, let me know if you ever change that device usage thing in hid.cpp that you mentioned in f1 23 post — I’d like to see if that helps the wheel get recognized more easily in some games.. 
thanks in advance!

ranenbg, 2025-07-11:
Hi man, thank you for testing, I'm glad that FFB is now sorted out in v250.

About endstop force, in v250 I made it in a way that it is activated just a little bit before the axis end, by 1/64 of axis range to be exact, from both sides. I did it intentionally, because when you set a pot as an input for FFB axis, in a joystick for example, it can only go up to the axis end, which would result in the absence of endstop force. With an optical encoder, it's possible to go far beyond the axis range, and I could make the endstop activate exactly at the end of the axis range. I understand that it is annoying, as before in v24X endstop was like this. I can make it back for 1-FFB axis and for 2-FFB axis firmware option, I can leave the small 1/64 gap, this is easy to do by using precompiler if statements, without extra memory cost. I'd like to keep one version of firmware, which has this kind of chameleon ability to transform into many colors.

 I'll make this change and then recompile all HEX and make a re-release soon.

closed 2025-07-11 by ranenbg