ranenbg/Arduino-FFB-wheel #103: xy shifter

Gandoun, 2025-07-16:
[image]

Why are the buttons not pressed in the game?

ranenbg, 2025-07-16:
Hi, tnx for reporting the issue with XY shifter in v250. I've just did a quick check of  HEX v250fx that you use and it does seem to have some problem with gear to button conversion. You came just in the right time as I'm in the process of fixing some other smaller bugs, I'm planning a re-release of v250.

I would appreciate if you could do some more testing and tell me which button gets reported for each XY shifter gear. I can see that 2nd gear is working, but I can't easily and quickly test the other ones at the moment. Help in this regard would be appreciated.

ranenbg, 2025-07-17:
Hi, it turns out this was a known bug in v250. I have already made a fix some time ago; however, I've recompiled all (35+) HEX files today. I will be doing a re-release of v250 sometime later today. You can then use the fixed HEX and it should work fine.