ranenbg/Arduino-FFB-wheel #105: Is it possible to use this esp32 as buttonbox on wheel wirelessly instead of NANO or promicro

Mamunmk1, 2025-07-30:
[image]

ranenbg, 2025-07-30:
Hi, yes, absolutely, just not with my firmware. There is already available Bluetooth firmware for ESP32wroom that turns it into a game device with buttons, it can even have axis as well, but it has no FFB. You can use it along side my firmware that will act as a wheel and this other board and firmware as a separate stand alone button box with many buttons, up to 64 I think.

closed 2025-07-30 by Mamunmk1

ranenbg, 2025-07-30:
Hi, you can have a look at this also, I did play with it for some time.
It's for Esp32Wroom, you can user Arduino IDE to upload it to ESP32 but you
need to install ESP boards first.