ranenbg/Arduino-FFB-wheel #107: I made a huge mistake.

AR9A, 2025-08-11:
So, I'll start with the main problem I caused.
 I looked at the motor driver wiring from another place and thought that for projects like this, the wiring is always the same. What I did was i connect R_EN and L_EN simultaneously and to digital pin 8 of the Arduino Leonardo, and RPWM to pin 10, LWPM to pin 9, VCC to 5 volts of the Arduino, and GND to GND of the Arduino. Also, the firmware version I used was v250htx. As soon as I connected the power supply (12v 30amp) to turn on the BTS7960 driver, a buzzing sound was heard from the power supply, and after about 2 minutes, the driver capacitor blew and smoke rose from the driver. After that, I changed the capacitor for the driver with same specs as orginal one and wired everything exactly according to your wiring. Now, when I test the force feedback, the motor (RS550) only turns in one direction, that is, only to the right, and if I try to turn it in the other direction, a sound like a buzzing sound is heard from both the driver and the power supply, meaning that the motor does not turn to the left at all. Also, when I was doing all these tests, only the bts7960 was connected to the Arduino.
My questions are: 
1- Is this driver damaged and do I need a new driver? 2- Has this action caused any problems for the Arduino board? (It is currently on and I can flash the firmware on it and the encoder works)
I would appreciate your guidance. Thank you for your time.

ranenbg, 2025-08-11:
Hi, yeah, that was a very silly mistake, but it happens. Make sure to carefuly read and follow my wiring instructions for h-bridge motor driver.

My suggestion is to right away order new arduino leonardo and bts7960 boards. There is a high chance that arduino is fine, so if budget is tight you may just get a new bts7960. I think one channel is probably gone. I'm not sure why this would happen, in "t" firmware pin8 is part of button matrix, it's configured as digital input or output, can't remember. In eaither case, motor would not spin in any direction at all, or would be very jerking in both directions.

To check if motor is fine, reverse its polarity and see if it rotates in the opposite direction normally, you can use 9V battery, but only power it for 1s or so. As I don't usualy make such wiring mistakes, I can't tell you what may have got damaged, you will have to find that out yourself.

AR9A, 2025-08-11:
Special thanks for your response.
I tested the motor and it turns in both directions with the battery and there is no problem.
So I will buy a new bts 7960 because my budget is limited and I will announce the result here

AR9A, 2025-08-17:
Hello again. The new driver arrived and I'm afraid to test it lol. I had a few questions. I would appreciate your advice. Is it better to have the PWM on Fast PWM or Phase-correct PWM? Do I need to modify the PWM sections at all? Also, the motor I want to use now is a windshield wiper motor. And which part of the Adreno FBB gui is related to FBB and reduces its intensity?

ranenbg, 2025-08-17:
Hi, do not touch anything in the firmware (code), you can use HEX that I compiled, or just compile yourself with ypur desired firmware options.

In GUI set PWM+- mode, you can use 8kHz or 16kHz pwm frequency and fast pwm type. Use general gain slider to set final FFB strenght.

closed 2025-08-17 by AR9A

ranenbg, 2025-08-24:
Yes, try with some min torque it will help with lower forces.

AR9A, 2025-08-24:
I installed the potentiometer correctly for the throttle alone, but the three pedals work at the same time. What is the problem? Thanks.

ranenbg, 2025-08-24:
The problem is that signal from lin A0 is reflected to other floating pins. You need to put other pins to gnd if you do not want them to pickup noise from other neigbouring pins.