ranenbg/Arduino-FFB-wheel #108: What firmware to install on Arduino Pro Micro

Stauzebench, 2025-08-12:
Hi Ranenbg,
I am very excited to build my Arduino FFB Wheel and want to ask what HEX I need. My Setup will be Pro Micro with Single BTS7960 and AS5600. I maybe want 3 pedals, but 2 are good for now with buttons for up and down shift. So just basic. There are so many files to choose and I dont want to make a mistake by choosing the wrong one. 
Thanks for your help.

ranenbg, 2025-08-12:
Hi, you can't make a mistake, as you can always flash another HEX. Don't be afraid to experiment. Go in proMicro folder v250 and find a HEX that has "w" in it, that one has as5600 support.

Stauzebench, 2025-08-13:
Alright thanks,
so I use either bcdwxm, or bcdwxgm. What are the differences with these two firmwares?

ranenbg, 2025-08-13:
The difference between those two is the "g" option, which is for analof FFB output via i2C dac, mcp4725 but you don't need it if you plan to use an h-bridge motor driver like bts7960. The option "b" stands for 2 axis FFB, which is meant to be used for flight controlers.

You may read a ver info txt file which is located in proMicro hex folder, I've put there description of all firmware options that each HEX contains.

Stauzebench, 2025-08-13:
Can you compile me a hex that can handle the following hardware.
One Axis for steering with AS5600 and Motor driver input with BTS7960.
I first want to make gas and breaking onto the wheel itself, something like the ps4/ps5 R2/L2 triggers. Is it possible to do that with pressure sensors or is a potentiometer more easy to begin with.
Two inputs for shifting up and down.
I think that is all to begin simple racing, if I haven't forgot something. I hope that is not too much to ask for.
Thanks in advance.

ranenbg, 2025-08-13:
You can use v250cdwhxm, I've already compiled that one and included it in latest release. It has everything you mentioned plus some extra features like hardware re-center button on pin D2 and a hat switch which takes up 4 normal pins for buttons. I'm not sure if there is enough button pins left for your other functions, but this is a good start, as this HEX is suitable for your gamepad build.

At some latter point I'll compile v250cdwxm which is the one you need.

Stauzebench, 2025-08-16:
Hey,
I had time to put everything together and flash the firmware to the arduino. At first all seems to work and I could feel some kind of force trom the motor. I opened the ffb iracing setup.exe and without any other settings i choose simple force and the motor began to spin endlessly in one direktion. To not damage something i unpluged the psu and after replugging, the motor couldnt begin to run, the psu always rebooted, like too much power drawn. I unwired the motor to get in the settings, but the arduino wasnt recognised anymore. After unwired everytning and testet around, I came across that the ibt is shorted at the positiv side. Is there some kind of protection like a diode we can use to protect the driver from inducting to much current back into? I use a MY1016 24V 350W and this thing will push back alot of current back into the system if nothing can absorb it.

ranenbg, 2025-08-16:
Hi, by default wheelcheck sets coefitient for simple spring effect to max, which is 10000. This means that even a tinyest movement of X axis will cause FFB output to max out. That motor is obviously too big of a bite for default bts7960 (without improved cooling).

I allways recommend to test FFB in my GUI. By default firmware has an autocenter spring activated. Even at 100% strenght it is much weaker than the one from wheelcheck. It will output max FFB only when you max out X-axis (or whichever axis is tied to ffb). You can use general gain slider to set overall FFB to an acceptable and safe level for your hardware combination.

Usualy, one would use a braking resistor circuit to dampen the energy from the motor when it is in generator mode. This is whenever no ffb is sent but motor is still spinning due to inertia or by external force from hand. Normaly for motors like RS775 or even RS875 with 12V psu this is not needed as bts7960 can handle the damping. But for 24V 350W motor, back elektromotor force is way to high. You can easily damage it if voltage goes above 30V or so. You can check with multimeter in voltag mode how much voltage your motor can generate in open circuit - just voltmeter connected.

Stauzebench, 2025-08-16:
Thanks for explaining.
Yeah, I have ordered some more BTS and will use your GUI to test FFB and connect just 12V for testing and setting up.

Stauzebench, 2025-08-16:
One more question.
I have an IBT-4 laying aroung. Is it possible to use this as well? It just has In1 and In2 to controll the motor. GPT says: 

IN1 = PWM, IN2 = LOW → Motor forward

IN1 = LOW, IN2 = PWM → Motor reverse

IN1 = LOW, IN2 = LOW → Motor coast (free spin)

IN1 = PWM, IN2 = PWM (both HIGH/PWM) → Motor brake (shorted, only use briefly).

closed 2025-10-07 by ranenbg