ranenbg/Arduino-FFB-wheel #109: PWM mode problem

CboatDesign, 2025-08-21:
Hello

I use Arduino promicro board and Cytron md20a controller, this type working with pwm+dir signals, firmware v240wxm, PWM Pin9, DIR Pin 16.
When I started the wheel control software the motor working fine but only one direction, when I turn it right the spring working well it turn the wheel for 0 position, but I turn it left direction the motor continuously rotate with full speed.
I read the whole document but I couldn't solve this.
Please help.

Thanks, Istvan.

ranenbg, 2025-08-21:
Hi, tnx for reporting this issue. It looks like that dir signal is missing for whatever reason. You should measure with a multimeter at your driver input side if this signal is being changed from 0V to 5V.

As you may know, I have recently made a fix for PWM signal generation for firmware v250, so PWM+DIR mode should work fine, but I didn't test it for proMicro yet. There may be a firmware problem, I'll check it. Did you try the HEX v250wxm from latest release?

ranenbg, 2025-08-22:
Just pay attention that firmware ending with "m" has replacement pinouts for proMicro board. It is possible to upload that firmware to leonardo board as well, but then dir signal will be missing as it has no pin 16.

CboatDesign, 2025-08-22:
Yes, you right, I put this one 250dwfx on the Leonardo, the wheel check see it, PWM 9 DIR 11 but the spring force working only the middle, after I turn some direction the motor rotation become to continiouosly both direction.

ranenbg, 2025-08-22:
Hi, that is good news. It seems that something is reversed; now you have FFB changing direction, but it's the opposite. The spring force is trying to make the wheel go away from the center, no matter which direction you turn it. Is that correct?

CboatDesign, 2025-08-22:
I changed the polairty and the wheel working fine, I dont understand why not work with the proMicro, but the Leonardo is good. 
Thanks for your help, you are was very helpful!
Are you have a patreon or something else?

ranenbg, 2025-08-22:
Great that you got it working on Leonardo. It should also work fine with proMicro, just use HEX from today's release.

I don't have patreon, tnx for asking, I do have paypal, you can see it in my gui.